4201VitruvianBots/ChargedUp2023 #199: All setpoints should auto-adjust as the robot moves, unless the operator manually moves the wrist/elevator, at which point the elevator/wrist will use their current positions as setpoints.

gavinskycastle, 2023-02-19:


gavinskycastle, 2023-02-22:
This still has not been done yet

jonathandao0, 2023-02-23:
Revised this to match current understanding of StateHandler.

AndrewKempen, 2023-02-23:
Why do we set limits for the wrist when it is in the stowed state? Shouldn't we just set the wrist to the stowed angle?

I think limits for the wrist should be set as a simple function of where the elevator is. When the elevator is below the threshold height, we should set the up and down limits to the below threshold limits and when the elevator is above the threshold we can expand the wrist limits.

AndrewKempen, 2023-02-23:
I think we also want to have states that control the transition of the elevator through the threshold point. We should have a LOW_TRANSIT and a HIGH_TRANSIT state that we enter whenever we aren't in a predefined wrist and elevator state. These states will have logic that looks at our desired state and appropriately attempts to move the elevator and four bar to the desired state unless it requires  going through the threshold point in which case it would block transition to the opposite TRANSIT state untill wrist and elevator have moved to angles that are safe for crossing the threshold height.

closed 2023-03-02 by gavinskycastle